ust-modules/USTSIPIN02 #5: Improvements for 5V supply

ChroustJan, 2022-08-23:
Upravit zapojení FTDI pro napájení 5V, je tam třeba pro variantu 5 V přidat jumper rozpojujíci VCC a VCCIO a volitelný odpor (pájecí jumper), který přemostí linearní stabilizátor na 3V3 v případě, kdy se neosadí stabilizátor.

kaklik, 2023-02-17:
Myslím si, že tahle záležitost takhle jednoduše vyřešit nejde, protože i současný návrh obsahuje konektory na kterých se pro jejich použití předpokládá 3V3 logika. 
V nové verzi to nevypadá, že by to mělo nějak výrazně změnit https://github.com/UniversalScientificTechnologies/LABDOS01/issues/6

V takovém kontextu mi snaha o převedení celého napájení na 5V nedává smysl.

kaklik, 2023-04-10:
Toto issue přestalo být se změnou #12  relevantní.

closed 2023-04-10 by kaklik